ballaswag/guppyscreen #2: IDEA

Zipciok, 2023-12-14:
Hi, i have idea. It would be nice to have one tab for macros that user can edit  for example like that 
[image]

Zipciok, 2023-12-15:
Thanks for fast response. I have question, thats my belt graph. what is ideal one? how much HZ should be? 110hz?
[image]

I just click start on this screen.
[image]

ballaswag, 2023-12-15:
See here https://github.com/Frix-x/klippain-shaketune/blob/main/docs/macros/belts_tuning.md I'm no expert at printer tuning.

Zipciok, 2023-12-15:
And one more thing. I would love to have file sorting system, for example by date of addition.

Zipciok, 2023-12-15:
Wow so fast replay. Thanks, that help alot

ballaswag, 2023-12-20:
[Screencast from 12-20-2023 12:07:25 PM.webm](https://github.com/ballaswag/guppyscreen/assets/145094472/2eea92e1-4a93-41a4-a161-8c56e3ac3fc6)
Expanded macros with parameters.

Zipciok, 2023-12-20:
> Screencast.from.12-20-2023.12.07.25.PM.mp4 
> Expanded macros with parameters.

That's cool, this is what i needed. There will be option to Pin 📌 few and have them displayed on the top of this list?

ballaswag, 2023-12-21:
yeah some kind of filtering and option to pin and unpin will be part of it.

Zipciok, 2023-12-21:
> yeah some kind of filtering and option to pin and unpin will be part of it.

Thanks, you are making realy good job!

closed 2023-12-26 by ballaswag