ballaswag/guppyscreen #4: Freeze in "print" screen

Horgul123, 2023-12-18:
When I try to open the list of the files on the printer it freezes. I waited several minutes, but nothing happens
[image]

ballaswag, 2023-12-18:
Can you run this endpoint against your moonraker and share me the json output. It lists all the gcode files and that’ll help me identify the issue.

https://moonraker.readthedocs.io/en/latest/web_api/#list-available-files

Horgul123, 2023-12-18:
hmmm.. Try to figure out how to do this.... Could you please give me a hint?

ballaswag, 2023-12-18:
Run this via ssh on your k1 `curl localhost:7125/server/files/list`. If you `guppyscreen.log` in `/usr/data/printer_data/logs`, please share that too.

Horgul123, 2023-12-18:
`root@K1-D35C /tmp [#] curl localhost:7125/server/files/list
{"result": [{"path": "3DBenchy.gcode", "modified": 1699424397.0, "size": 2979584                                                                                                             , "permissions": "rw"}, {"path": "600S-TEST-7m.gcode", "modified": 1699424397.0,                                                                                                              "size": 2497441, "permissions": "rw"}, {"path": "Amazing_2934379_PLA_27m34s.gco                                                                                                             de", "modified": 1701872131.8151731, "size": 3526327, "permissions": "rw"}, {"pa                                                                                                             th": "Amazing_2940397_PLA_23m8s.gcode", "modified": 1701936043.563049, "size": 2                                                                                                             994766, "permissions": "rw"}, {"path": "base_PLA_6h19m.gcode", "modified": 17021                                                                                                             21221.8473463, "size": 57206484, "permissions": "rw"}, {"path": "baumkugel v1_PL                                                                                                             A_1h27m.gcode", "modified": 1701690708.9867065, "size": 11042940, "permissions":                                                                                                              "rw"}, {"path": "baumkugel v1_PLA_44m40s.gcode", "modified": 1701629197.8964353                                                                                                             , "size": 4422241, "permissions": "rw"}, {"path": "body_PLA_1h42m.gcode", "modif                                                                                                             ied": 1702194561.595437, "size": 16481994, "permissions": "rw"}, {"path": "Cat.g                                                                                                             code", "modified": 1699424397.0, "size": 4845908, "permissions": "rw"}, {"path":                                                                                                              "collar_PLA_2m57s.gcode", "modified": 1702482199.6674147, "size": 174114, "perm                                                                                                             issions": "rw"}, {"path": "Creality-filament-sensor-coupler-lever_PLA_1h2m.gcode                                                                                                             ", "modified": 1701754391.4948206, "size": 4122918, "permissions": "rw"}, {"path                                                                                                             ": "Creality-filament-sensor-coupler-lever_PLA_23m10s.gcode", "modified": 170172                                                                                                             5406.7322192, "size": 1582589, "permissions": "rw"}, {"path": "DoorSeal_BackPlat                                                                                                             e_3mmFoam_PLA_24m6s.gcode", "modified": 1701760810.371018, "size": 765542, "perm                                                                                                             issions": "rw"}, {"path": "Feuerwehr Platzchenform_PLA_1h20m.gcode", "modified":                                                                                                              1701444363.0753825, "size": 7979719, "permissions": "rw"}, {"path": "Feuerwehr                                                                                                              Platzchenform_PLA_1h8m.gcode", "modified": 1701432414.5387244, "size": 7466795,                                                                                                              "permissions": "rw"}, {"path": "Feuerwehr Platzchenform_PLA_5h19m.gcode", "modif                                                                                                             ied": 1701460201.853762, "size": 31790389, "permissions": "rw"}, {"path": "Filam                                                                                                             ent Splitter 4-Way_PLA_1h41m.gcode", "modified": 1702560827.290111, "size": 1354                                                                                                             1301, "permissions": "rw"}, {"path": "Flow Test 5mm_PLA_18m56s.gcode", "modified                                                                                                             ": 1701702679.166862, "size": 506772, "permissions": "rw"}, {"path": "Harry_Pott                                                                                                             er_Wand_PLA_33m35s.gcode", "modified": 1702214712.5221012, "size": 5063605, "per                                                                                                             missions": "rw"}, {"path": "head_PLA_1h20m.gcode", "modified": 1702486277.214482                                                                                                             8, "size": 9221932, "permissions": "rw"}, {"path": "house_bottom_PLA_4h48m.gcode                                                                    [... cut ...]        ": "Zero_Eyes_PLA_2m9s.gcode", "modified": 1702736759.4230132, "size": 264157, "                                                                                                             permissions": "rw"}, {"path": "Zero_Eyes_PLA_54s.gcode", "modified": 1702737734.                                                                                                             root@K1-D35C /tmp [#]
`

[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/13707836/guppyscreen.log)

eustacefan, 2023-12-18:
I have same problem.
server/files/list: [file_list.txt](https://github.com/ballaswag/guppyscreen/files/13707924/file_list.txt)
[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/13707928/guppyscreen.log)

ballaswag, 2023-12-18:
Thanks for the report and logs. Will take a look.

ballaswag, 2023-12-19:
@eustacefan @Horgul123 just released 0.0.14-beta, can you try if that still freezes for you. Upgrade the same way you install ```sh -c "$(wget --no-check-certificate -qO - https://raw.githubusercontent.com/ballaswag/guppyscreen/main/installer.sh)"```

Also added file sorting y date created/az, layer count in print status.

MrDix, 2023-12-19:
Sorting by date now works perfectly out of the box and can be switched to other orders easily. Thank you so much for the immediate implementation!

Horgul123, 2023-12-19:
@ballaswag Will test it tomorrow when the current print is finished.

eustacefan, 2023-12-20:
@ballaswag It work fine, thanks

Horgul123, 2023-12-20:
@ballaswag : Thanks a lot fro the fix. No freeze and thumbnails are displayed :)

Because I'm curious: What was the problem? I think, you tested this without issues.

ballaswag, 2023-12-20:
my guess is one of the metadata field is not present and i wasn’t checking before accessing it. I reworked in some checks and also removed some sketchy new and delete. Good to see the issue resolved to you.