ballaswag/guppyscreen #5: Hotend heated in "extrude" screen when selecting temperature

Horgul123, 2023-12-18:
Not sure if it's a bug or a feature:

In the screen where I could extrude / retract I have the values for speed, length and temperatur.
Length and speed can be changed and nothing happens -> fine
When I select a temperature the printer starts heating the hotend, which I wouldn't expect. For me the temperature is a value like speed and length and not a command.

I would expect that the printer starts heating as soon as I want to extrude or retract. Like it's working in the creality frontend.
Beeing able to have a default would be great.
So the workflow would be: go to extrude screen, check if the three values are ok, and just click extract or retract.

And a small question, because i'm still a beginner:
The retract command just retracts or (like in the creality frontend) extrudes a little bit and than starts retracting? (and where can I configure this? Didn't find a macro)

ballaswag, 2023-12-18:
The extrude and retract panel is meant for unloading and reloading. The underlining are using `G1 F<speed> E<amount>`. The temperature selector is supposed to allow fine control on what temperature you want to extrude and retract at. I think I can reverse this and also select a default, then pressing the button will start the action.

Creality has custom macros for these actions. You can find them in their gcode_macos.cfg. The extrude before retract is probably coded only those custom macros.

With guppy screen you can still use these macros via the console. Mapping macros to a custom button is in the works.

Horgul123, 2023-12-18:
Thanks for the explanation.
This means:
I need to heat the nozzle and when it reaches the temperature I press extrude/retract?

I think having a constistent behaviour for the three selectors would give a better user experience

Horgul123, 2023-12-29:
Did you fix this issue, too?
With the latest beta I can change the temperature and it starts heating when I click on extrude and not after selecting a temperature

ballaswag, 2023-12-29:
yup changed to not heat temp is selected, but only heat when extrude/retract is pressed.

Horgul123, 2023-12-29:
Perfect! Thanks a lot 😃👍

closed 2023-12-29 by Horgul123