ballaswag/guppyscreen #6: IDEAS #2

d4vex, 2023-12-19:
Hello, thanks for this great major improvement. Have few ideas to improve it even more as i'm using it for few days now and missing some functions.

1. Extrusion would be great to have more lenghts as I barely use 5mm, 10mm or 15mm, maximum of 35mm is not enough for me to change color, so 50mm or 100mm would be better. Also I'm missing cooldown extruder button in temperature section.
2. Global cooldown button on home screen.
3. Easier selection of actuall printing status. Now, if I'm not missing something, is needed to go to the print section and then to the status.
4. Better gcode files sorting, ideal is to have newest files on top.
5. Generally font size could be larger.

David

Horgul123, 2023-12-19:
+1 for the first 2 points.

In addition: A kind of progress/success message for extrude/retract. Tried it today and wasn't sure if it got my command or not

ballaswag, 2023-12-19:
Thanks for the suggestions. 4 is done and will be in next release. It enables file sorting by date modified and alphabetical. I’m new to 3d printing as well, so still learning the workflows as I’m building out Guppy Screen.

VioSynthax, 2024-02-10:
Is support for vertically oriented touchscreens feasible?

Herbi88, 2024-02-15:
Hi, thx for your great work. Would it be possible to make USB Stick work on GuppyScreen? Sometimes I have no connection and it would be nice if we could use USB Stick for printing files. Thx

Michoo66, 2024-02-16:
Hi,
it would be nice to have bigger font for temperature values on the main screen (nozzle, bed, chamber). Thank you for your hard work.

d4vex, 2024-05-11:
I installed version 0.26 after some time using creality screen. And wow what a progress. Great work. Is it possible to add let say to guppyconfig.json configuration for extrusion/retraction lenghts as these 1-35mm is not usable for me. So i could place lenghts i want. 50, 70 etc. Thanks for your work.