ballaswag/guppyscreen #7: Feature Request: Screen Auto Off

WubberSnipes, 2023-12-20:
Love this Project! What an improvement over what Creality offers out of the box. 
Upon my initial testing I have noticed that the screen no longer shuts off after a couple minutes and as my printers are left on 24/7 a configurable Screen Auto Off feature similar to what Creality had on the original screen would be valuable.

Thank you for your work!

ballaswag, 2023-12-20:
I think the default screen off timer is 5mins or inactivity (no touches). It’s currently only configurable by editing the guppyconfig.json. Let me know if that’s not the case and share the guppyscreen.log.

WubberSnipes, 2023-12-21:
Thanks for the info. I found the guppyconfig.json under /usr/data/guppyscreen but have yet to make any modifications to it. Looks like the default screen timeout is 600 seconds and when first booted up the screen does sleep after that interval however, after further testing the issue I have found that after the initial sleep, the screen is turning its self on and off about every hour for some reason. I will attach the log for you to check out. 

For the large majority of this log I have not been interacting with the printer or its display whatsoever. I do however keep the klipper/moonraker websocket and sometimes SSH terminal open on my computer so maybe there is some sort of keep-alive check triggering every hour somehow also triggering the screen. I don't know though, would be curious to know what you think.

[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/13741889/guppyscreen.log)
This is the behavior I am finding:
[2023-12-20 16:08:27.497] [guppyscreen] [debug] putting display to sleeping
[2023-12-20 17:10:02.461] [guppyscreen] [debug] waking up display
[2023-12-20 17:20:02.463] [guppyscreen] [debug] putting display to sleeping
[2023-12-20 18:21:37.432] [guppyscreen] [debug] waking up display
[2023-12-20 18:31:37.434] [guppyscreen] [debug] putting display to sleeping
[2023-12-20 19:33:12.397] [guppyscreen] [debug] waking up display
[2023-12-20 19:43:12.398] [guppyscreen] [debug] putting display to sleeping
[2023-12-20 20:44:47.366] [guppyscreen] [debug] waking up display
[2023-12-20 20:54:47.367] [guppyscreen] [debug] putting display to sleeping
[2023-12-20 21:56:22.333] [guppyscreen] [debug] waking up display
[2023-12-20 22:06:22.336] [guppyscreen] [debug] putting display to sleeping
[2023-12-20 23:07:57.302] [guppyscreen] [debug] waking up display
[2023-12-20 23:17:57.304] [guppyscreen] [debug] putting display to sleeping
[2023-12-21 00:19:32.269] [guppyscreen] [debug] waking up display
[2023-12-21 00:29:32.271] [guppyscreen] [debug] putting display to sleeping
[2023-12-21 01:31:07.232] [guppyscreen] [debug] waking up display
[2023-12-21 01:41:07.235] [guppyscreen] [debug] putting display to sleeping
[2023-12-21 02:42:42.199] [guppyscreen] [debug] waking up display
[2023-12-21 02:52:42.205] [guppyscreen] [debug] putting display to sleeping
[2023-12-21 03:54:17.169] [guppyscreen] [debug] waking up display
[2023-12-21 04:11:32.592] [guppyscreen] [debug] putting display to sleeping
[2023-12-21 05:13:07.559] [guppyscreen] [debug] waking up display

ballaswag, 2023-12-21:
Thanks for the detail report. Will take a look, I do recall another user reporting screen turning back on without a touch. In terms of turning off on boot, that’s due to the delay of the clock syncing on boot (ntpd). The sleep timer in guppy screen currently relies on the system clock for timing informanton.

Hammerfest, 2024-03-24:
This is also happening on both my K1 and K1 Max

TheWebMachine, 2024-03-24:
I can confirm I have this issue on 1.3.3.8 with K1C. I didn't think I did at first, but I just walked into the print closet and saw the screen on. I haven't touched it in about 24hrs.