ballaswag/guppyscreen #8: 0.0.14-beta: Files list is only updated when klipper gets restarted

MrDix, 2023-12-20:
I've updated to your 0.0.14-beta release a few minutes after you've released it as I'm one of the requesters (I've asked you on redit about it) of the new files sorting option which you have introduced with this release. 

After the upgrade I had to power cycled the printer because Klipper reported a MCU error directly after the Guppy Screen installer script finished. This might have happend because I forgot that the printer was currently printing while I executed the installer and answered the question if it is ok to to restart Klipper now with 'yes'. 
Guppy Screen and Klipper where up and running after the reboot. GCode files are shown in the files list after pressing the 'Print' button and can be sorted as preferred. Thank you very much!

After uploading new GCode files via mainsail the they directly show up in mainsails file manager but do not show up in the files list after clicking the 'Print' button on Guppy Screen. 
I could not find a way to make Guppy Screen refresh the files list other than restarting klipper (from within the screen menu).

Mainsail file manager: 
[image]

Guppy Screen:
[image]
Files uploaded (and printed) within the last hours are not included in the GCode file choice. 

Guppy Screen (after Klipper restart):
[image]
All files are available in files menu on Guppy Screen as well.

Is there maybe a directory listing cached for the newly introduced sorting and the cached file list is not refreshed when new files were added to this directory or when the print menu got reopened? 

Hope this is an easy to fix problem for you!

ballaswag, 2023-12-20:
Thanks for the detail report. I added a Refresh button, next to the sort buttons. Does that not work at reloading for new files?

It’s a manual refresh, I’ll have to see if moonraker publish an event for newly added files, if so then I guess that can be automatic. Please let me know if the refresh button works or not.

ballaswag, 2023-12-20:
Also, you can always start the print in another client like mainsail. Guppy screen will automatically pick up the status and show the job status.

MrDix, 2023-12-20:
I can't believe I was so blind to not seeing the refresh button in the upper left corner 🙁
Sorry for opening this ticket over here!

An implicit refresh when reentering the print menu would be a great feature in case there is no moonraker event which can be used to trigger the automatic refresh after a file got uploaded. Just in case others are as blind as me 🤪

The printing via mainsail works of course without issues and I've used this with the current guppy screen multiple times. It is just very common for me that when I create a multi-part CAD model that I push the plates one by one to the printer and than only print the plates one by one just using the screen.

The ticket here can be closed as the solution already existed (refresh button) before I was asking for it.

closed 2023-12-21 by MrDix