ballaswag/guppyscreen #11: Feature request: work flow for changing filament during print

Horgul123, 2023-12-22:
Maybe I've just overseen it:

I tried a two color print which requires filament change after a specific layer.
The printer got the PAUSE command and guppyscreen offered me to cancel or resume the print.

To unload the old filament and to load the new one I needed to go to the extruder screen.
After changing the filament I didn't find an option to resume the print.

In mainsail I could continue.

Is the resume option after a filament change missing or did I just not found it? (think I've checked all menues 🤔) 

Would expect to see at the home screen that a print is running and just paused.

ballaswag, 2023-12-22:
if you press Print, there should be a Status button if a printer is active. From there you will get back into the print job page where you’ll get the resume button.

something I’m working on:
1. Add unload and load buttons to the extrude panel. This allows the user to map their own unload and load macro to these 2 buttons. For now K1 will be mapped to wuit_material, and load_material respectively. Users have the option to remap in guppyconfig.json.
2. making an active print more obvious, probably adding a small print progress to the main panel so it’s easier to get back into the job status panel.

ballaswag, 2023-12-23:
See if release 0.0.15-beat does what you want. The extrude panel now adds unload/load (these are mapped to K1's custom macros LOAD_MATERIAL, QUIT_MATERIAL). You can also manually map them to something else by editing the guppyconfig.json.

Horgul123, 2023-12-23:
Thanks a lot! :)
will test it during the day. Really great, that you listen to us users 👍

Horgul123, 2023-12-23:
Noticed another small thing. In the print screen you show 3 fans. I wondered, that all are at 0%. Compared with mailsail I saw, that these are fans 1-3, but the hotend fan is not displayed Maybe this fan could be added?

[image]
[image]

Horgul123, 2023-12-23:
@ballaswag Load/unload works perfect! Thanks a lot!

ballaswag, 2023-12-23:
> Noticed another small thing. In the print screen you show 3 fans. I wondered, that all are at 0%. Compared with mailsail I saw, that these are fans 1-3, but the hotend fan is not displayed Maybe this fan could be added?
> 
> ![2023-12-23_13-40-46](https://private-user-images.githubusercontent.com/45976162/292627674-7f05a466-0f87-4227-b9b6-e86903109ca0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzNTc5MTMsIm5iZiI6MTcwMzM1NzYxMywicGF0aCI6Ii80NTk3NjE2Mi8yOTI2Mjc2NzQtN2YwNWE0NjYtMGY4Ny00MjI3LWI5YjYtZTg2OTAzMTA5Y2EwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIzVDE4NTMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjMjMzM2QwNmFlOTE4Y2IyOWMxNzJjNzdmMTgzYjE0ZWY2ZDY0NDMxY2FlMjdiM2FjZjcwMjhkYTRkOTM5MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IVqY5g9JfxFYIjv9mQA9v4y_Js8m5tOCkR_g9uywLdM) ![WhatsApp Image 2023-12-23 at 14 07 30](https://private-user-images.githubusercontent.com/45976162/292627676-83ffbc69-613c-431c-8e13-0f00bdd123a6.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzNTc5MTMsIm5iZiI6MTcwMzM1NzYxMywicGF0aCI6Ii80NTk3NjE2Mi8yOTI2Mjc2NzYtODNmZmJjNjktNjEzYy00MzFjLThlMTMtMGYwMGJkZDEyM2E2LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIyM1QxODUzMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYjVkYTM3NjM0NmFmYjAxYTRmNjU5Yzg0OWNiODAzMjg4ODVhNDQ3OGEyNzM0NTlmMGU2NmUxODliYWVhNzAwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D34DTakiiQ5uT3JGTP1J7fcBW-E2tmpzyj5-yHPF5ms)

The k1 defines the fans as output pins and that’s what you see in mainsail. I think fan1 is the toolhead fan. I will need to install mainsail and verify. Not sure why it would report 100% in mainsail.

Horgul123, 2023-12-23:
Not sure... In my screenshot you can 3 fans with 0%. The hotend fan seems to be another fan

closed 2024-04-14 by ballaswag