ballaswag/guppyscreen #13: Feature request: Heating off in Extrude screen

Horgul123, 2023-12-25:
In the extrude screen you can select different temperatures and the current temperature ist displayed.
To turn the heating off you need to go back to the previous screen.

Idea 1: 
Add "0" to the list of temperature presets

Idea 2:
Open temperature setting screen when clicking on the extruder temperature i the upper left corner (the scrren which is opened when changinf the temperature in the main screen)

ballaswag, 2023-12-25:
Will think about this. Maybe slider is better than the selector for the extrude panel. Might add a cooldown button under the load/unload buttons too.

Horgul123, 2023-12-26:
Sounds great! :)

ballaswag, 2023-12-29:
Cooldown button added in 0.0.16-beta

closed 2023-12-29 by ballaswag

Horgul123, 2023-12-29:
It's working! :)
But I didn't see, that he accepted my cooldown order.
Idea: Change the temoerature display so that it shows current and target temperature
e.g.

Extrude 50° / 240°

or something like this. Then you can see when he should start or end the heating.