ballaswag/guppyscreen #16: Install of both screen is failing

perincolony1, 2023-12-29:
Installation of both types failing
[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/13796005/guppyscreen.log)

Same message both screens types

waiting for printer to initialize , its up on my pc no issue

Let me know
[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/13796064/guppyscreen.log)

ballaswag, 2023-12-29:
[guppyscreen.tar.gz](https://github.com/ballaswag/guppyscreen/files/13796419/guppyscreen.tar.gz)
Can replace and try this `guppyscreen` in /usr/data/guppyscreen.

perincolony1, 2023-12-29:
***@***.*** /usr/data/guppyscreen [#] ./guppyscreen
[2023-12-30 07:09:01.896] [guppyscreen] [debug] registering new method
notify_active_spool_set, handler SpoolmanPanel
[2023-12-30 07:09:01.908] [guppyscreen] [debug] registering new method
notify_gcode_response, handler ConsolePanel
[2023-12-30 07:09:01.983] [guppyscreen] [debug] found existing resp_method
notify_gcode_response with handlers, updating handler callback
InputShaperPanel
[2023-12-30 07:09:01.991] [guppyscreen] [debug] found existing resp_method
notify_gcode_response with handlers, updating handler callback
BeltsCalibrationPanel
[2023-12-30 07:09:02.019] [guppyscreen] [info] connecting to printer at
ws://127.0.0.1:7125/websocket
[2023-12-30 07:09:02.022] [guppyscreen] [debug] websocket connecting
[2023-12-30 07:09:02.046] [guppyscreen] [debug] onopen
[2023-12-30 07:09:02.047] [guppyscreen] [debug] init panel connected
[2023-12-30 07:09:02.049] [guppyscreen] [debug] send_jsonrpc:
{"id":0,"jsonrpc":"2.0","method":"printer.objects.list"}
[2023-12-30 07:09:02.095] [guppyscreen] [debug] onmessage(type=text
len=2056): {"jsonrpc": "2.0", "result": {"objects": ["webhooks",
"configfile", "mcu", "mcu nozzle_mcu", "mcu leveling_mcu", "mcu rpi",
"gcode_macro xyz_ready", "gcode_macro _IF_HOME_Z", "gcode_macro
_IF_MOVE_XY", "gcode_macro _HOME_X", "gcode_macro _HOME_Y", "gcode_macro
_HOME_Z", "gcode_macro PRINTER_PARAM", "gcode_macro AUTOTUNE_SHAPERS",
"gcode_macro LOAD_MATERIAL_CLOSE_FAN2", "gcode_macro
LOAD_MATERIAL_RESTORE_FAN2", "gcode_macro SET_E_MIN_CURRENT", "gcode_macro
RESTORE_E_CURRENT", "gcode_macro LOAD_MATERIAL", "gcode_macro
QUIT_MATERIAL", "gcode_macro Qmode", "gcode_macro Qmode_exit", "gcode_macro
M204", "gcode_macro M205", "gcode_macro M106", "gcode_macro M107",
"gcode_macro M900", "gcode_macro WAIT_TEMP_START", "gcode_macro
WAIT_TEMP_END", "gcode_macro PRINT_CALIBRATION", "gcode_macro
FIRST_FLOOR_PAUSE_POSITION", "gcode_macro ACCURATE_G28", "gcode_macro
START_PRINT", "gcode_macro PRINT_PREPARED", "gcode_macro
PRINT_PREPARE_CLEAR", "gcode_macro END_PRINT_POINT_WITHOUT_LIFTING",
"gcode_macro END_PRINT_POINT", "gcode_macro END_PRINT", "gcode_macro
FIRST_FLOOR_PAUSE", "gcode_macro FIRST_FLOOR_RESUME", "gcode_macro PAUSE",
"gcode_macro INPUTSHAPER", "gcode_macro BEDPID", "gcode_macro
TUNOFFINPUTSHAPER", "gcode_macro RESUME", "gcode_macro CANCEL_PRINT",
"gcode_macro G29", "calibrate_shaper_config", "gcode_macro GUPPY_SHAPERS",
"gcode_macro GUPPY_BELTS_SHAPER_CALIBRATION", "gcode_macro
GUPPY_EXCITATE_AXIS_AT_FREQ", "gcode_move", "print_stats", "fan_feedback",
"custom_macro", "idle_timeout", "virtual_sdcard", "heaters",
"temperature_sensor mcu_temp", "temperature_sensor chamber_temp", "tmc2209
stepper_x", "tmc2209 stepper_y", "tmc2209 stepper_z", "tmc2209 extruder",
"heater_bed", "pause_resume", "filament_switch_sensor filament_sensor",
"filament_switch_sensor filament_sensor_2", "heater_fan hotend_fan",
"output_pin fan0", "output_pin fan1", "output_pin fan2", "output_pin LED",
"probe", "bed_mesh", "display_status", "exclude_object", "motion_report",
"query_endstops", "system_stats", "manual_probe", "toolhead", "extruder"]},
"id": 0}
[2023-12-30 07:09:02.099] [guppyscreen] [debug] send_jsonrpc:
{"id":1,"jsonrpc":"2.0","method":"server.files.roots"}
[2023-12-30 07:09:02.101] [guppyscreen] [debug] send_jsonrpc:
{"id":2,"jsonrpc":"2.0","method":"server.database.get_item","params":{"key":"console","namespace":"fluidd"}}
[2023-12-30 07:09:02.103] [guppyscreen] [debug] send_jsonrpc:
{"id":3,"jsonrpc":"2.0","method":"server.database.get_item","params":{"namespace":"guppyscreen"}}
[2023-12-30 07:09:02.105] [guppyscreen] [debug] send_jsonrpc:
{"id":4,"jsonrpc":"2.0","method":"printer.gcode.help"}
[2023-12-30 07:09:02.107] [guppyscreen] [debug] send_jsonrpc:
{"id":5,"jsonrpc":"2.0","method":"server.files.list","params":{"root":"gcodes"}}
[2023-12-30 07:09:02.109] [guppyscreen] [debug] send_jsonrpc:
{"id":6,"jsonrpc":"2.0","method":"server.info"}
[2023-12-30 07:09:02.110] [guppyscreen] [debug] found user configured
extruder extruder
[2023-12-30 07:09:02.112] [guppyscreen] [debug] found user configured
heater temperature_sensor chamber_temp
[2023-12-30 07:09:02.113] [guppyscreen] [debug] found user configured
heater heater_bed
[2023-12-30 07:09:02.117] [guppyscreen] [debug] sensor cb registered name
extruder, cont 0x76305200, this 0x76304424, np 0x7f95bc30
[... 880 lines cut ...]
"sense_resistor": "0.100", "stealthchop_threshold": "0", "uart_address":
"3", "diag_pin": "^PB12", "driver_sgthrs": "65"}, "stepper_y": {"step_pin":
"PB8", "dir_pin": "!PB7", "enable_pin": "!PC3", "microsteps": "32",
"rotation_distance": "72", "endstop_pin":
"tmc2209_stepper_y:virtual_endstop", "position_endstop": "-0.5",
"position_min": "-0.5", "position_max": "226", "homing_speed": "36",
"homing_retract_dist": "0"}, "tmc2209 stepper_y": {"uart_pin": "PA10",
"interpolate": "True", "run_current": "1.5", "hold_current": "1.0",
"sense_resistor": "0.100", "stealthchop_threshold": "0", "uart_address":
"3", "diag_pin": "^PB13", "driver_sgthrs": "65"}, "stepper_z": {"step_pin":
"PB6", "dir_pin": "!PB5", "enable_pin": "!PC3", "microsteps": "16",
"rotation_distance": "8", "gear_ratio": "64:20", "endstop_pin":
"tmc2209_stepper_z:virtual_endstop", "position_endstop": "0",
"position_max": "255", "position_min": "-5"}, "tmc2209 stepper_z":
{"uart_pin": "PA11", "uart_address": "3", "run_current": "0.8", "diag_pin":
"^PB14", "stealthchop_threshold": "0", "sense_resistor": "0.100",
[... cut ...]
"max_extrude_cross_section": "80", "step_pin": "nozzle_mcu:PB1", "dir_pin":
"nozzle_mcu:PB0", "enable_pin": "!nozzle_mcu:PB2", "microsteps": "16",
"rotation_distance": "6.9", "nozzle_diameter": "0.400",
"filament_diameter": "1.750", "heater_pin": "nozzle_mcu:PB7",
"sensor_type": "EPCOS 100K B57560G104F", "sensor_pin": "nozzle_mcu:PA0",
"pressure_advance": "0.04", "pressure_advance_smooth_time": "0.040",
"min_temp": "0", "max_temp": "320", "control": "pid", "pid_kp": "23.818",
"pid_ki": "1.846", "p

perincolony1, 2023-12-29:
Ok here is the log file

And i did check somehow octoeverywhere is working but i can not find it in
any directory

Here is guppy log

On Fri, Dec 29, 2023 at 5:10 PM Per Backman ***@***.***> wrote:

> ***@***.*** /usr/data/guppyscreen [#] ./guppyscreen
> [2023-12-30 07:09:01.896] [guppyscreen] [debug] registering new method
> notify_active_spool_set, handler SpoolmanPanel
> [2023-12-30 07:09:01.908] [...]

perincolony1, 2023-12-30:
hopefully now

perincolony1, 2023-12-30:
[guppyscreen (1).log](https://github.com/ballaswag/guppyscreen/files/13796710/guppyscreen.1.log)

i add it here directly

perincolony1, 2023-12-30:
2 new log files
1 is from after i finished a print

2 is from after i reverted to stock and then install with zbolt
[guppyscreen (3).log](https://github.com/ballaswag/guppyscreen/files/13798743/guppyscreen.3.log)
[guppyscreen (2).log](https://github.com/ballaswag/guppyscreen/files/13798744/guppyscreen.2.log)

ballaswag, 2023-12-30:
Thanks for the logs. I think the issue is related to width and height for thumbnails being returned as strings instead of numbers. I’m haven’t look to see why moonrakers metadata parser would do that sometimes and not other. One work around is to see if a rescan of the file will update width and height to numbers.

https://moonraker.readthedocs.io/en/latest/web_api/#scan-gcode-metadata

perincolony1, 2023-12-30:
Send full comand and i do it

perincolony1, 2023-12-30:
example of gcode file
[Micron_Cowls.zip](https://github.com/ballaswag/guppyscreen/files/13798942/Micron_Cowls.zip)

perincolony1, 2023-12-30:
here is the list log
[list.log](https://github.com/ballaswag/guppyscreen/files/13798981/list.log)

ballaswag, 2023-12-31:
[guppyscreen.tar.gz](https://github.com/ballaswag/guppyscreen/files/13800645/guppyscreen.tar.gz)
@perincolony1 see if this fixes your issue. Replace the guppyscreen binary in /usr/data/guppyscreen with the one in this tarball. This one checks for a string representation of a number before converting for thumbnails width/height.

perincolony1, 2023-12-31:
Happy new year

Unfortunately same error on the K1 screen waiting to initialize 

Here is guppy log
[guppyscreen (4).log](https://github.com/ballaswag/guppyscreen/files/13801909/guppyscreen.4.log)

perincolony1, 2023-12-31:
Happy new year

Same screen error "waiting for printer to initialize"

I uploaded new log to git

Per

ballaswag, 2024-01-01:
@perincolony1 seems like you’re not the only user with the issue, see here https://github.com/Arksine/moonraker/issues/760

perincolony1, 2024-01-01:
> only user with the issue, see here Arksine/moonraker#760

Ohh interesting

perincolony1, 2024-01-03:
Here is new guppyscreen log file

perincolony1, 2024-01-03:
root@K1-CF09 /usr/data/guppyscreen [#] ./installer.sh
=== Installing Guppy Screen ===
Checking for a working Moonraker
 Found klipper path: /usr/share/klipper
 Found config dir: /usr/data/printer_data/config
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 1134k  100 1134k    0     0   494k      0  0:00:02  0:00:02 --:--:--  713k
 Test starting Guppy Screen
13150 root      0:00 /usr/data/guppyscreen/guppyscreen
 Guppy Screen started sucessfully, continuing with installation
Setting up Guppy Macros
printer.cfg already includes GuppyScreen cfgs
 Backing up original K1 files
=== Do you want to disable all Creality services (revertable) with GuppyScreen installation? ===
  Pros: Frees up system resources on your K1 for critical services such as Klipper (Recommended)
  Cons: Disabling all Creality services breaks Creality Cloud/Creality Slider.

Disable all Creality Services? (y/n): n

Setting up GuppyScreen
Replacing mathplotlib ft2font module for plotting PSD graphs
Restart Klipper now to pick up the new changes (y/n): y

Restarting Klipper
no need to update custom config files
old_version: 1023
new_version: 1023
Starting GuppyScreen
13518 root      0:00 /usr/data/guppyscreen/guppyscreen
 Successfully installed Guppy Screen. Enjoy!
root@K1-CF09 /usr/data/guppyscreen [#] curl -XPOST “http://127.0.0.1:7125/server/files/metascan?filename=corner-v2-K1_0.4_Hyper ABS_1h59m.gcode”
{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/usr/data/moonraker/moonraker-env/lib/python3.8/site-packages/tornado/web.py\", line 1681, in _execute\n    result = self.prepare()\n\n  File \"/usr/data/moonraker/moonraker/moonraker/app.py\", line 965, in prepare\n    raise tornado.web.HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}root@K1-CF09 /usr/data/guppyscreen [#]

perincolony1, 2024-01-04:
[guppyscreen new.log](https://github.com/ballaswag/guppyscreen/files/13824501/guppyscreen.new.log)
Last log file

ballaswag, 2024-01-05:
This is fixed in 0.0.17-beta. You can get zbolt themed version now.

closed 2024-01-05 by ballaswag

perincolony1, 2024-01-05:
Thanks, I'll check it out!

Will do later

Per

perincolony1, 2024-01-05:
I dont see a difference between them

perincolony1, 2024-01-05:
Its not looking like octoscreen at all, looks the same to me

ballaswag, 2024-01-05:
The only difference are the icons, the panels are all the same.