ballaswag/guppyscreen #21: GuppyScreen blocks ssh/printer

Horgul123, 2024-01-07:
Some time ago I installed GuppyScreen at my K1 using your installer, The FW on the K1 was the version, whoch was the current version 4 weeks ago.
Everything worked well.

Yesterday I resetet my K1, installed the FW 1.3.3.3.3 and the Guilouz Helper Script, which now supports the installation of GuppyScreen.
When I started the first print I noticed, that I can't reach my printer via ssh anymore and after the print I noticed, that GuppyScreen is no longer working at the K1.
After power off/on everythings working again.

Guilouz mentioned, that I should try to restart GuppyScreen via GUPPY_K1_SSH_RESTART macro, but I can't find this macro.

Question:
Could my problem caused by GuppyScreen?
If yes: Is it related to the installation via the helper script?
Any ideas to solve this problem? I really like GuppyScreen, so I it would be very sad if it's causing problems.

ballaswag, 2024-01-07:
Did you lose SSH after restarting even with Guppy Screen working? See if you can run the following to restart ssh via a console (on guppy, fluidd, mainsail)

`RUN_SHELL_COMMAND=GUPPY_K1_SSH_RESTART PARAMS=restart`

Then try ssh. If that doesn't work you can try opening your printer.cfg via fluidd/mainsail add this after the [include]s:
```
[gcode_shell_command SSH_RESTART]
command: /etc/init.d/S50dropbear
timeout: 600.0
verbose: True
```

Save and restart Klipper. Then via a console run:
```
RUN_SHELL_COMMAND=SSH_RESTART PARAMS=restart
```

All of the above expects you to have the gcode shell extension installed. Guppy Screen should have installed that for you. Once you get back SSH, I suggest you run the install command again from the main page.

Horgul123, 2024-01-07:
After restarting the K1 everything worked again.
Tried to restart GuppyScreen as a test:
`14:03:19
{"code":"key61, "msg":"Unknown command:RUN_SHELL_COMMAND=", "values": ["RUN_SHELL_COMMAND="]}
14:03:17
RUN_SHELL_COMMAND=GUPPY_K1_SSH_RESTART PARAMS=restart`

Tried to uninstall/reintall it via helper script and uninstall via helper scrips and installing it with your installation command, but still getting the same error message.

Any ideas?

ballaswag, 2024-01-07:
Are your other gcode shell commands working? Gcode shell is this file

https://github.com/ballaswag/guppyscreen/blob/main/k1/k1_mods/gcode_shell_command.py

installed in /usr/share/klipper/klippy/extra.

GUPPY_K1_SSH_RESTART is from guppy_cmd.cfg installed in /usr/data/printer_data/config/GuppyScreen and included in your printer.cfg. Make sure these are present and restart klipper, then you should get back that specific guppy restart macro.

Horgul123, 2024-01-07:
OK. Thanks.... All the files and the include in the printer cfg is there....
Will restart the printer after the current print.
At the moment, ssh ist working fine.

Horgul123, 2024-01-08:
The restart via command is still not working.
I don't use any shell commands, so I don't know if they are working.

The Python script is there:
[image]

In the printer.cfg you files are included:
[image]

your cfg is there:
[image]

and the command is in the cfg
[image]

@ballaswag Any idea why it's still not working?

Horgul123, 2024-01-08:
Do you knpw how I can check if the Klipper Gcode Shell Command is installed?

ballaswag, 2024-01-08:
looks installed. Did you restart klipper after installing and how are you running the command?

should be

`RUN_SHELL_COMMAND=GUPPY_K1_SSH_RESTART PARAMS=restart`

Horgul123, 2024-01-08:
Restarted the complete printer several times, but it's not working. Looks like the gcode shell command is not installed:
19:24:55
{"code":"key61, "msg":"Unknown command:RUN_SHELL_COMMAND=", "values": ["RUN_SHELL_COMMAND="]}
19:24:55
RUN_SHELL_COMMAND=GUPPY_K1_SSH_RESTART PARAMS=restart

But I don't know how to check or reinstall it. Your script doesn't offer a new installation

ballaswag, 2024-01-08:
Oops sorry, the command is actually 

`RUN_SHELL_COMMAND CMD=GUPPY_K1_SSH_RESTART PARAMS=restart`

note the extra CMD=

Horgul123, 2024-01-08:
YES!  That's the problem :)  THANKS!

closed 2024-01-08 by Horgul123