ballaswag/guppyscreen #22: Change default extrude temps

bcutty, 2024-01-07:
I mainly use the screen to load and unload but would like to raise the temps in the default screen, any way to do this currently?

ballaswag, 2024-01-07:
You can use the Extrude panel, it has temp selection, cooldown, load/unload which you can map to your custom macros. They map to LOAD_MATERIAL/QUIT_MATERIAL (on the k1/max by default). You can remap them by editing the guppyconfig.json file and restarting guppy screen.

bcutty, 2024-01-07:
i see the extrude panel, the max temp there is 240. even when i set the temp higher before going in, it goes down to whatever temp is selected there before extruding or retracting so i'm trying to figure out how to raise that 240 max default

ballaswag, 2024-01-07:
For load and unload, the temp is configured in the macros via then macro parameter printer.custom_macro.default_extruder_temp. Will fix this so that it uses the selected temp. For now you probably need to change that parameter in the cfg file to the desired temp.

also in the todo is to make those selector customizable, so users can define their own number ranges.

closed 2024-01-31 by bcutty