ballaswag/guppyscreen #23: K1 Screen bootloop

evgarthub, 2024-01-10:
Last time I used my printer I reinstalled guppyscreen to 0.0.17.
Today I turned on printer and it shows loading ("Creality") screen infinitely. Printer itself works, I can still connect to it.
Can you share steps to how get logs of what's happening?

evgarthub, 2024-01-10:
I rebooted it couple of times - didn't help.
I Just updated moonraker using mainsail ui come to printer and now see guppyscreen is back. 
This issue still can be relevant to debugging steps for next time.

ballaswag, 2024-01-10:
Guppy Screen logs are in /usr/data/printer_data/logs/guppyscreen.log. Should be able to grab it off of fluidd/mainsail file browser.

Sometimes the guppy log doesn’t capture the actual error. If guppy screen crashes and you have access to ssh, you can start guppyscreen on the cli and that’ll show the actual error.

run `/usr/data/guppyscreen/guppyscreen` via ssh and check the log dumped to terminal.

evgarthub, 2024-01-14:
apparently it takes forever to load, it's not a bootloop. 
This what I've got by running `/usr/data/guppyscreen/guppyscreen`
[image]

evgarthub, 2024-01-14:
And this is log file:
[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/13932641/guppyscreen.log)