ballaswag/guppyscreen #24: New wireless auto connection

javirub, 2024-01-12:
When i connected with the oem screen interface to a wireless connection, it gets saved and autoconnect to that network everytime you power on-off creality k1 printer.

With guppy screen, the previous saved network still reconnecting, but when you add a new one, you have to connect manually everytime you restart the machine.

ballaswag, 2024-01-12:
Thanks for the report. Issue most likely due to guppy not issuing a save_config against wpa supplicant after switching networks. Will take a look at fixing for next release.

javirub, 2024-01-12:
> Thanks for the report. Issue most likely due to guppy not issuing a save_config against wpa supplicant after switching networks. Will take a look at fixing for next release.

Yes, anyway for some reason, the /etc/wpa_supplicant.conf (it should be in /etc/wpa_supplicant/wpa_supplicant.conf) has no stored data inside, put manually my wireless data and still not working. Finally i made it work with wpa_cli

closed 2024-04-14 by ballaswag