ballaswag/guppyscreen #25: Force Login

halTWO3, 2024-01-13:
Force Login is set True in Moonraker.conf
guppyscreen.conf updated: 
moonraker_api_key: false, (to) moonraker_api_key: "<api key>",

Screen is stuck at initializing.

guppyscreen.log:

[2024-01-02 20:51:02.421] [guppyscreen] [debug] registering new method notify_active_spool_set, handler SpoolmanPanel
[2024-01-02 20:51:02.458] [guppyscreen] [debug] registering new method notify_gcode_response, handler ConsolePanel
[2024-01-02 20:51:02.526] [guppyscreen] [debug] found existing resp_method notify_gcode_response with handlers, updating handler callback InputShaperPanel
[2024-01-02 20:51:02.532] [guppyscreen] [debug] found existing resp_method notify_gcode_response with handlers, updating handler callback BeltsCalibrationPanel
[2024-01-02 20:51:02.562] [guppyscreen] [info] connecting to printer at ws://127.0.0.1:7125/websocket
[2024-01-02 20:51:02.564] [guppyscreen] [debug] websocket connecting
[2024-01-02 20:51:02.654] [guppyscreen] [debug] onopen
[2024-01-02 20:51:02.655] [guppyscreen] [debug] init panel connected
[2024-01-02 20:51:02.656] [guppyscreen] [debug] send_jsonrpc: {"id":0,"jsonrpc":"2.0","method":"printer.objects.list"}
[2024-01-02 20:51:02.675] [guppyscreen] [debug] send_jsonrpc: {"id":1,"jsonrpc":"2.0","method":"server.files.roots"}
[2024-01-02 20:51:02.677] [guppyscreen] [debug] send_jsonrpc: {"id":2,"jsonrpc":"2.0","method":"server.database.get_item","params":{"key":"console","namespace":"fluidd"}}
[2024-01-02 20:51:02.678] [guppyscreen] [debug] send_jsonrpc: {"id":3,"jsonrpc":"2.0","method":"server.database.get_item","params":{"namespace":"guppyscreen"}}
[2024-01-02 20:51:02.679] [guppyscreen] [debug] send_jsonrpc: {"id":4,"jsonrpc":"2.0","method":"printer.gcode.help"}
[2024-01-02 20:51:02.680] [guppyscreen] [debug] send_jsonrpc: {"id":5,"jsonrpc":"2.0","method":"server.files.list","params":{"root":"gcodes"}}
[2024-01-02 20:51:02.682] [guppyscreen] [debug] send_jsonrpc: {"id":6,"jsonrpc":"2.0","method":"server.info"}
[2024-01-02 20:51:02.723] [guppyscreen] [debug] server_info {"error":{"code":-32602,"message":"Unauthorized"},"id":6,"jsonrpc":"2.0"}

Maybe I'm missing something, thanks in advance.

ballaswag, 2024-01-13:
moonraker api key is currently not implemented in guppy screen yet. Will add support in a future release.

halTWO3, 2024-01-13:
Thanks for the very quick reply. Looking forward for that release, till then, will just use trusted clients filter.  Thanks again for this amazing mod.  Cheers!

closed 2024-01-13 by halTWO3