ballaswag/guppyscreen #27: Pause and cancel button gone

neturmel, 2024-01-18:
in 0.0.19-beta, on smallscreen, when printing, there is a progress bar but no pause and cancel button anymore-

ballaswag, 2024-01-18:
oh no, let me fix. I must have verified in the small resolution. Changed something for larger screen and forgot to double check m.

ballaswag, 2024-01-18:
I see the buttons on my simulator for smaller screens. Can you share a picture of what you're seeing.
[image]

neturmel, 2024-01-19:
[image]

I updated from 0.0.18 (if that matters)

ballaswag, 2024-01-22:
I can’t reproduce on the simulator and I don’t have the actual hardware. I tried to redo the alignment in 0.0.20-beta latest version. Please let know if that resolves the issue if you get a chance to update.

neturmel, 2024-01-22:
I solved it! Updated to 0.0.20-beta to no avail. But than I realized, that the buttons were indeed there, until the peview image was rendered. Them I realized, I sliced with Orca and a preview image size of 32x32, 300x300 (because this worked better in creality screen), may be this was the problem? Tried now with 96x96, 300x300 and all is well.

closed 2024-01-25 by neturmel