ballaswag/guppyscreen #29: Screen freeze on K1 mid print

tarodofchaos, 2024-01-19:
The screen freezes in the middel of a print and it won't respond to touch unless you restart the service or manually power cycle the printer.

This happens with a V2 Creality K1

- Firmware version 1.3.3.3
- KAMP-for-K1-Series v1.1.2-8-g9e942708
- mainsail v2.9.1
- moonraker v0.8.0-262

Attached guppy log.
[guppyLog.txt](https://github.com/ballaswag/guppyscreen/files/13986531/guppyLog.txt)

mattskent, 2024-01-20:
Having the same issue here

ballaswag, 2024-01-20:
Thanks for the report. Going to try two things. In the short term if I can track down where the crash happens, I’ll provide some auto restart so guppy screen will just automatically start up again. In the long term, will track down the crash and fix the issue.

ballaswag, 2024-01-22:
Latest 0.0.20-beta automatically restarts guppy screen if it dies. If you get a chance to update, please let me know if it works better.

mattskent, 2024-01-31:
Just confirming I haven't had any screen freezes so far on the latest version

ballaswag, 2024-02-02:
Closing this since the latest version will auto respawn if guppy screen is terminated.

closed 2024-02-02 by ballaswag