ballaswag/guppyscreen #30: Guppy Screen Kills Camera During Prints

hjkeller, 2024-01-24:
When printing with Guppy Screen installed on a K1Max, Firmware 1.3.3.5, the mcu rpi uses 100% capacity and stops streaming images.
I removed guppy screen(only change) and the issue has not occured. Which log/information can I provide to help confirm?

ballaswag, 2024-01-24:
please attach guppyscreen.log from /usr/data/printer_data/logs/guppyscreen.log

hjkeller, 2024-01-24:
> please attach guppyscreen.log from /usr/data/printer_data/logs/guppyscreen.log

added to original post.

ballaswag, 2024-01-24:
Also can you provide me a timeline. When the streaming start and when it terminated. Just an approximation will help me narrow where to look.

hjkeller, 2024-01-24:
1: Freshly boot printer Streaming is okay and webcam is active
2: Start Print with KAMP, camera stays active through calibration and first layers, moonraker timelapse is capturing frames. It has failed in the first 10-15% of long prints. So the first hour of an 8+ hour print.
3: Print continues normally, refreshing mainsail view in OrcaSlicer, Firefox on desktop, and chromium based browser on mobile.
4: At somepoint here the camera stops and MCU RPI pegs to 100%
5: Print finishes, timelapse doesn't render
6: Reboot printer, camera stream returns.

I ran some shorter prints, benchy and some 12 minute prints and the stream doesn't fail, but timelapse doesn't render them so I think that is unrelated to the actual stream.

closed 2024-01-25 by brightandshinyobject

hjkeller, 2024-01-25:
This issue appears unrelated to guppy screen as it occurs without guppy screen installed.