ballaswag/guppyscreen #31: [FR] Add HappyHare integration

Rai-Oliveira, 2024-01-26:
It would be nice to have HappyHare integration in GuppyScreen, like the KlipperScreen HappyHare edition fork. 
This is currently the biggest hurdle in my attempt to add a MMU to the K1 ecosystem. Although not mandatory, this integration provides great user experience, which in my opinion is always a main goal.

https://github.com/moggieuk/KlipperScreen-Happy-Hare-Edition/tree/master
Link for reference, just in case.

ballaswag, 2024-01-26:
Hey, I was made aware of this extension a few days ago by another user. The author has done an awesome job. At this time I don't have the bandwidth and hardware to test the implementation. I'll mark this as a todo down the roadmap.

flyespresso, 2024-05-19:
I'm also interested in this, it'd be huge to be integrated native as right now as @Rai-Oliveira linked it's an add-on...

I'd be happy to test with my DoomCube V2 + ERCF V2 + BTT 5" Screen

k1-801, 2024-06-28:
Also interested in this, I might invest some time soon to make it work with my ERCF v1.1

flyespresso, 2024-07-14:
Any chance you've taken a look at this @ballaswag ?

k1-801, 2024-07-17:
> Also interested in this, I might invest some time soon to make it work with my ERCF v1.1

Finished installing HappyHare on my K1 Max, checking how the HappyHare integration works in klipperscreen