ballaswag/guppyscreen #32: [FR] Moonraker Authentication support

skullbone, 2024-01-27:
It would be nice to be able to use guppy with forced logins enabled

ballaswag, 2024-01-28:
Will take a look at how that works.

mattskent, 2024-02-01:
+1 for this request, even just an option to set the API key in a config file somewhere and then pass that in to moonraker would be sufficient.

halTWO3, 2024-02-03:
Yep, was inquiring about it a few weeks ago.

Liionboy, 2024-04-13:
any news? Because is not secure to let moonraker without API Key

briankelly, 2025-01-06:
+1 on this. i was confused because i saw the `moonraker_api_key` var in guppyconfig.json, but it doesn't look like it ever gets used. 

```
[2025-01-06 14:38:37.869] [guppyscreen] [debug] send_jsonrpc: {"id":7,"jsonrpc":"2.0","method":"server.info"}
[2025-01-06 14:38:37.894] [guppyscreen] [debug] server_info {"error":{"code":-32602,"message":"Unauthorized"},"id":7,"jsonrpc":"2.0"
```

dimagord, 2025-05-11:
Still no auth support?