ballaswag/guppyscreen #36: Unable to select/see spoolman

ItsDeidara, 2024-02-04:
I'm sure it's user error but as of right now I am unable to use spoolman after installing Guppyscreen.

It is currently grayed out and not able to be tapped on the touch screen itself, and "invisible" within Fluidd. I am able to see the option of Spoolman within my layout editor, but even enabled it does not seem to appear.

Is there a setting I missed that I was supposed to setup?

ballaswag, 2024-02-04:
Did you enable spoolman in moonraker? See https://moonraker.readthedocs.io/en/latest/configuration/#spoolman

ItsDeidara, 2024-02-04:
I did not enable it moonraker, good catch! I just did, and did a full reboot afterwards. Unfortunately it did not make a difference. It is not showing up in Fluidd and the direct URL within moonraker does not seem to work

closed 2024-04-14 by ballaswag