ballaswag/guppyscreen #39: Stuck in the Creality Screen

Chelejeff, 2024-02-07:
Hello Guys,

I installed the Guppyscreen around two weeks ago.
I decided to uninstall it last night, following the instructions provided.
After a reboot, the screen is stuck on the Creality logo. However, I have access to Fluid and can control the printer from there. 
I do not have access to SSH, which complicates everything. I tried from Mac and Win. 
Would you happen to have any idea what to do?
Attached are the logs. 

[moonraker.log](https://github.com/ballaswag/guppyscreen/files/14200752/moonraker.log)
[guppyscreen.log](https://github.com/ballaswag/guppyscreen/files/14200755/guppyscreen.log)

avilshadow, 2024-02-27:
In my case it is not stuck, but print pause/cancel was not working..

Btw, how did you installed guppy without ssh?

baliboy79, 2024-06-04:
> Hello Guys,
>
> I installed the Guppyscreen around two weeks ago. I decided to uninstall it last night, following the instructions provided. After a reboot, the screen is stuck on the Creality logo. [...]

Hi. I have a same problem. My printer is Ender 3 V3. Creality logo on the screen and ssh access is not working. You fixed yours?

Liionboy, 2024-06-04:
Hello,

You can fix this only reflash nebula pad, i was in this situation and after reflash everything work normal

Thanks