ballaswag/guppyscreen #41: UI Not User Friendly

Scalegj, 2024-02-17:
I find Guppy Screen's UI to not be as user friendly as the original. Could you make a version that looks the same as the original but includes new features like spoolman.

Rulevoid, 2024-02-18:
The biggest issue for me is there's never like a throbber icon to show you that Guppy heard what you asked for and is doing it. For example when selecting anything under extrude (unload, extrude, retract) you have to just watch for other signs that it's doing what you want like watching the extruder temperature for example. Speaking of, if you select unload, it will never drop the extruder temp and will leave it hot. Another issue is with Load. When I select that, it just keeps pushing out filament forever until I turn off the printer. I assume that's a bug too. Also the extruder temperature is 3 characters but Guppy wraps the text at 2 characters. Lots of fit and finish needed, but a good start!

ballaswag, 2024-02-18:
Will work on UX in terms of hints when something is happening. Maybe like a spinner on the buttons when it's in action.

The load and unload are macros that came with the KE. They are mapped to LOAD_MATERIAL and QUIT_MATERIAL respectively. If you need to change those behavior you can update those macros or you can remap the two buttons to your custom routines. The mappings are editable in guppyconfig.json under "default_macros".

YAYABOCIANA, 2024-02-23:
@ballaswag how to edit guppyconfig.json ? im running Guppy on my KE and id like to change mappings and screen rotation but I can't find .json file at all    I know it is inside .tar file but im unable to update guppy using modified tar file

ballaswag, 2024-02-23:
> @ballaswag how to edit guppyconfig.json ? im running Guppy on my KE and id like to change mappings and screen rotation but I can't find .json file at all    I know it is inside .tar file but im unable to update guppy using modified tar file

There really isn't a functional mapping. The KE will only look nice in landscape mode. If you installed Guppy Screen with the install command from this GitHub it should automatically set it in portrait mode.

If you really want to try guppyconfig.json is under /usr/data/guppyscreen. There's a display_rotation attribute. Valid values are 0 to 3, 90 degrees each increment.

henrivdr, 2024-03-21:
> The biggest issue for me is there's never like a throbber icon to show you that Guppy heard what you asked for and is doing it. For example when selecting anything under extrude (unload, extrude, retract) you have to just watch for other signs that it's doing what you want like watching the extruder temperature for example. Speaking of, if you select unload, it will never drop the extruder temp and will leave it hot. Another issue is with Load. When I select that, it just keeps pushing out filament forever until I turn off the printer. I assume that's a bug too. Also the extruder temperature is 3 characters but Guppy wraps the text at 2 characters. Lots of fit and finish needed, but a good start!
> 
> Edit - I'm using an Ender 3 V3 KE Nebula Pad.

@ballaswag  I noticed the temperature wrapping as well, on a CR-10 SE. I recall it was on 204C, and other values.