ballaswag/guppyscreen #49: initialize printer stuck

KennyVB, 2024-02-28:
suddely the screen is stuck in startup. printer works, everythjing but the screen works
[https://pastebin.com/0VwryU7v](url)

KennyVB, 2024-02-29:
`organic_PLA_2h16m-300x300.png","size":11504,"width":300}]
[2024-02-29 07:51:57.898] [guppyscreen] [debug] running locally, skipping thumbnail downloads
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be string, but is null`

ballaswag, 2024-02-29:
Thanks for the log. Have you upgraded moonraker and spoolman? This looks reproducible since it keeps on crashing during initialization.

KennyVB, 2024-03-02:
moonraker version v0.8.0-318-g9813011d
Spoolman is latest, as i tried to reinstall it, hopeing it would help

KennyVB, 2024-03-02:
i uninstalled with the script guppyscreen.
and as i told on discord: i set the thumbnails to standard size.

now i can't install guppyscreen, it stops/error :
Guppyscreen failed to start

KennyVB, 2024-03-02:
got it working agian. now it suddeny went trhough the installation.
log her:
[https://pastebin.com/LKnH3YMG](https://pastebin.com/LKnH3YMG)

ballaswag, 2024-03-02:
See if it works by disabling spoolman by removing the config section from moonraker.

KennyVB, 2024-03-02:
> See if it works by disabling spoolman by removing the config section from moonraker.

that... worked. 
i did check spoolman log on the server side, and it recived connection fine, and it has worked for some time. no updates, just suddely dead guppy :(

adec, 2024-03-19:
Interestingly I've had the same issue too on my Ender 3 V3 KE.  When spoolman is enabled in moonraker config, I also have the same errors with guppyscreen. Commenting out the spoolman config, restarting moon raker and then restarting guppyscreen makes things spring back into action again.

Version info:
Guppy Screen Version: 0.0.21-beta
moonraker: v0.8.0-324-ga3e4dac5
Spoolman version: 0.16.1

ballaswag, 2024-03-19:
Spoolman has a pretty big upgrade since my integration with it. Will take a look at fixing the integration with the newer spoolman versions.

adec, 2024-03-20:
Thanks. Let me know if I can do anything to help.

adec, 2024-03-20:
@ballaswag Is there any way to temporarily disable Spoolman support in the Guppy Screen configuration file to allow spoolman to continue to use it for tracking filament while waiting for the Guppy Screen support to be updated?

ballaswag, 2024-03-20:
Unfortunately not currently. If moonraker is configured with spoolman, guppy screen currently always try to enable it.

adec, 2024-03-20:
No problem. I'll probably fork the code and see if I can help resolving things.  I'm no cross-compilation expert unfortunately, and spdlog cross-compilation is currently putting up a fight. I'll initially add a disable-spoolman config option for now until I get time to work out what's wrong with the spoolman integration.

ballaswag, 2024-03-20:
Take a look at https://github.com/ballaswag/guppyscreen/pull/57, this PR has more detail on getting the build cross compiled. I have been working on another project, GuppyFLO https://github.com/ballaswag/guppyflo, and haven’t had time to update guppy screen.  But I welcome PRs for any of these projects.

adec, 2024-03-21:
I've identified the problem with the spoolman integration. It looks like the "material" attribute is missing in the output of the server.spoolman.proxy /api/v1/spool?allow_archived=true request.  I got the following output back from spoolman:

```
"jsonrpc": "2.0", "result": [{"id": 1, "registered": "2024-03-18T22:35:48Z", "first_used": "2023-12-25T
22:35:29.565000Z", "last_used": "2024-03-19T21:58:51Z", "filament": {"id": 1, "registered": "2024-03-18T22:35:06Z", "name": "Creality Hyper PLA - Black", "vendor": {"id": 1, "registered": "202
4-03-18T22:32:07Z", "name": "Creality", "extra": {}}, "density": 1.25, "diameter": 1.75, "weight": 1000.0, "settings_extruder_temp": 220, "settings_bed_temp": 60, "color_hex": "060606", "extra
": {}}, "price": 21.0, "remaining_weight": 986.9044794197093, "used_weight": 13.095520580290641, "remaining_length": 328245.7626314301, "used_length": 4355.58783000012, "archived": false, "ext
ra": {}}], "id": 10}
```

I notice in the latest version of spoolman, the material attribute is shown on the filament screen but the form for entering/editing filaments doesn't appear to prompt for it.  I've confirmed this is definitely the cause by tweaking spoolman_panel.cpp not to look for the material field - and it works properly.

Before I work on a proper fix and raise a pull request, I'll check out the spoolman repo to see whether the missing material field is intentional or a bug.

ballaswag, 2024-03-21:
Thanks for looking into this! That would do it. I use nlohmann/json for handling these request response. For any mistype or access to missing attributes, it throws an exception which crashes guppy screen. I can do a better job doing exception handling for this use case, but I really don't want to start adding exceptions handling code.

I tried to do existence validation before access in most places and sometimes get sloppy about that.

adec, 2024-03-22:
My pleasure. I have an admission to make. I managed to find the Material field in spoolman but in my haste I must of missed it as it's not a mandatory field. I'll check out what other attributes are optional which could cause issues and have a go at adding some validation for these fields.

ballaswag, 2024-04-14:
Closing as latest stable release should be checking for these instead of crashing. Please reopen if that's not the case.

closed 2024-04-14 by ballaswag