ballaswag/guppyscreen #50: Can't go back to status screen

Unwrapped3, 2024-03-02:
Whenever I leave the print status screen there is no way to come back, leaving me with Jo way to control the print from the touch screen. Sometimes I get a small icon up top on the home screen that a print is ongoing, sometimes this icon is also missing. However, when I do have it and press it, the input menu for extruder temperature pops up instead of opening the print status screen. 

Mainsail version: 0.8.0-319gf0ae5179
Guppy version: 0.0.21-beta 

Would appreciate any help!

ballaswag, 2024-03-02:
You can click print again and there should be a status button to get back to the print process.

Unwrapped3, 2024-03-02:
> You can click print again and there should be a status button to get back to the print process.

[image]

There is the "view job" button when I press Print but it does not work: nothing happens when pressed.

ballaswag, 2024-03-02:
Get out of that prompt and click the status button.

Unwrapped3, 2024-03-02:
> Get out of that prompt and click the status button. The mini status in the main panel is probably bugged.

Aha, yes, it works. Thanks a lot! 

Btw I'm not sure if I should open an another issue but the Z-Offset status is shown incorrectly in the print status menu if you've adjusted it and saved the config. 

At the same, in the Fine Tune menu it is correct though, leading to two different values being shown.

closed 2024-03-02 by Unwrapped3