ballaswag/guppyscreen #51: Layer counts on print screen not updating when printing from slicer

ajs123, 2024-03-06:
With SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count] in the print start gcode, 
and SET_PRINT_STATS_INFO CURRENT_LAYER={layer_num + 1} at each layer...

1. Power up printer (or restart firmware from Guppyscreen) and print from the slicer: Layer number and total display as expected.
2. Print a second print from the slicer: Layer number and total indicate the last layer from the prior print.
3. Print something from the Guppyscreen menu: Layer number and total display as expected.

sigmashig, 2024-03-11:
Vote for this issue. I have the same problem.

PJW58, 2024-04-14:
+1 - Same issue here.  I am currently resetting Guppy Screen between each print job to compensate.

ballaswag, 2024-04-14:
Try latest release. Layer count resets should have been fixed.

SSPW, 2024-04-17:
I tried 0.0.23-beta today.
Still no layer info on print status panel.
When i look at the changelog for the release, i dont see any changes for layers, only flow.