ballaswag/guppyscreen #52: Documentation/Make Bug? : Cross Compile Fails

elminster, 2024-03-07:
Following the instructions for a cross-compile;

```
Mipsel (Ingenic X2000E) - specific to the K1 SoC

Building for the K1/Max
1. unset SIMULATION && export CROSS_COMPILE=mips-linux-gnu-
2. make wpaclean && make wpaclient
3. make libhvclean && make libhv.a
4. make spdlogclean && make libspdlog.a
5. make clean && make -j$(nproc)
The executable is ./build/bin/guppyscreen
```

It fails on step 5 with:
` skipping incompatible spdlog/build/libspdlog.a `

Checking the archives it is not cross compiling libspdlog.a files
```
file color_sinks.cpp.o
color_sinks.cpp.o: ELF 64-bit LSB relocatable, x86-64, version 1 (GNU/Linux), not stripped
```

Whereas if I extract and check another archive like libhv.a they look fine
```
file appletls.o
appletls.o: ELF 32-bit LSB relocatable, MIPS, MIPS32 rel2 version 1 (SYSV), not stripped
```

elminster, 2024-03-09:
I did get it to compile by adding the following to the spdlog CMakeLists.txt, but have yet to test it

```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR mips)
set(tools /tmp/mips-gcc720-glibc229)
set(CMAKE_CXX_COMPILER "${tools}/bin/mips-linux-gnu-c++")
set(CMAKE_AR "${tools}/bin/mips-linux-gnu-ar")
```

ajs123, 2024-03-09:
I have the same error. Where in CMakeLists.txt did you place those lines?

ajs123, 2024-03-09:
I can confirm that the binary runs on the Nebula Pad when compiled this way, with the specified lines at the top.

ballaswag, 2024-03-09:
spdlog used to be build with their headers only options. I have since changed it to build and link as s static library. To cross compile spdlog, make sure your cross compile toolchain is in your PATH, then export CC and CXX to point a your cross compile toolchain. For example I do the following to build spdlog.a for mipsel:

```
export CC=mips-linux-gnu-gcc
export CXX=mips-linux-gnu-g++
cd spdlog && rm -rf build && mkdir build && cd build && cmake .. && make -j
```

ballaswag, 2024-03-09:
> I can confirm that the binary runs on the Nebula Pad when compiled this way, with the specified lines at the top.

The *-smallscreen.tar.gz variable from this repository's releases should work for the nebula pad. See assets here https://github.com/ballaswag/guppyscreen/releases

elminster, 2024-03-10:
Thanks. It did work on the K1c in the way I compiled it above when I tested it, but I will use try your method so as not to hack around with cmakefiles when I don't need to.

ajs123, 2024-03-10:
Thanks for this.

Can you advise what the build flags are to compile for the Nebula Pad? I’ve
defined GUPPY_SMALL_SCREEN which provides scaled down icons but the fonts
remain larger.

ballaswag, 2024-03-10:
For the nebula pad, I export these:

```
export CROSS_COMPILE=mips-linux-gnu-
export CC=mips-linux-gnu-gcc
export CXX=mips-linux-gnu-g++
export GUPPY_SMALL_SCREEN=1
export EVDEV_CALIBRATE=1
```

For the K1C, the same build for k1/max should work.

ajs123, 2024-03-10:
Cross-compile works with unmodified CMake config and the exports of CC and CXX and specified above. Thank you.

However, with GUPPY_SMALL_SCREEN=1, as shown by the two images, icons are scaled but text is not. 

[image]
[image]

The new binary is also 5.7 MB instead of the 3.5 MB of the binary from the distribution, which seems like much more than one would expect from a difference in font files.

ballaswag, 2024-03-10:
You can strip the binary for a smaller executable. With the smallscreen flag it should select the smaller font. Will check my setup later to verify.

ballaswag, 2024-03-10:
@ajs123 Since you're looking into the small screen version for nebula pad. One thing that'll be useful is the touch input calibration enabled for it. I don't own a nebula pad so I never figured out the min/max x/y for this display. So some users are reporting incorrect touches on their nebula pad. If you're up for it, see if you can figure out the correct min/max x/y for this device.

ajs123, 2024-03-10:
I'm happy to do that. Letme know where it's set, and maybe one of the
logging options is useful to see what taps at the corners give.

ballaswag, 2024-03-10:
Here's the function that handles touch https://github.com/ballaswag/guppyscreen/blob/e755e589e9e8b5798bb38d9ec75426047ca95ddc/src/main.cpp#L280

You should be able to add logs in there for the touch coordinates. Maybe log the raw coordinates before the calibration is done so you can find min/max then readjust those in the calibration function.

ajs123, 2024-03-10:
I put the coordinate logging in main.cpp (got confounded by the #includes
for logging in evdev.c). As best as I could slide a pencil off of the bezel
to hit the absolute corners,

Upper Left: 22, 20
Lower Left: 23, 225
Lower Right: 479, 239
Upper Right: 450, 23

Assuming the screen is mounted straight in the bezel, the ranges are
(22...479), (20..239).

I hope that helps. If not, I can look to getting the raw coordinates.

ballaswag, 2024-03-10:
Thanks for this, I don't have the display to test. You can use the value you discovered and replace the min/max x/y here https://github.com/lvgl/lv_drivers/blob/71830257710f430b6d8d1c324f89f2eab52488f1/indev/evdev.c#L222

Give it a try to see if it's better than the existing release. If so, those are the value we should use for the nebula pad and not the ones hardcoded now.

ajs123, 2024-03-11:
Cal is done and looks good. I moved it to another Issue thread since this one now has three issues in it. Re the fonts, just to tie up that loose end here, changing both lines as follows has no apparent effect, so the font must be getting set elsewhere.

```
lv_theme_t * th = height <= 480
      ? lv_theme_default_init(NULL, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED), true, &lv_font_montserrat_12)
      : lv_theme_default_init(NULL, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED), true, &lv_font_montserrat_16);
    lv_disp_set_theme(disp, th);
```

pellcorp, 2024-03-22:
you can also just make a change to the Makefile thus:

```
diff --git a/Makefile b/Makefile
index 1fce29d..4f78f40 100644
--- a/Makefile
+++ b/Makefile
@@ -102,7 +102,7 @@ libhv.a:
 
 libspdlog.a:
        @mkdir -p $(SPDLOG_DIR)/build
-       @cmake -B $(SPDLOG_DIR)/build -S $(SPDLOG_DIR)/
+       @cmake -B $(SPDLOG_DIR)/build -S $(SPDLOG_DIR)/ -DCMAKE_CXX_COMPILER=$(CXX)
        $(MAKE) -C $(SPDLOG_DIR)/build -j$(nproc)
```