ballaswag/guppyscreen #54: Nebula touch screen calibration

ajs123, 2024-03-11:
By adding code to log coordinates, I determined a good calibration for my Nebula pad (KE). On my unit, these result in X=0 at the very lefthand edge and Y=271 at the very bottom edge.

In lv_drv_conf.h:

`#  if EVDEV_CALIBRATE
#    define EVDEV_HOR_MIN         190             /*to invert axis swap EVDEV_XXX_MIN by EVDEV_XXX_MAX*/
#    define EVDEV_HOR_MAX      3850            /*"evtest" Linux tool can help to get the correct calibraion values>*/
#    define EVDEV_VER_MIN         300
#    define EVDEV_VER_MAX      3749 
#  endif  /*EVDEV_CALIBRATE*/
#endif  /*USE_EVDEV*/`

moriceh, 2024-03-14:
+1 on this, it would be very nice if we could change those values in a config file without having to rebuild the entire project, too :)

Liionboy, 2024-03-25:
Create a PR with this to be implement in next version

jktikka, 2024-04-27:
I am affected by this to the point of not being able to use Guppyscreen. 

Would love to see a fix for someone with less linux knowledge.

ballaswag, 2024-04-27:
@jktikka calibration is released in the latest build. Give that a try. It should fix the touch inputs on a nebula pad.

jktikka, 2024-04-27:
@ballaswag Is that using the commands on in the readme? I tried current and nightly and still no good unfortunatly.