ballaswag/guppyscreen #55: Setting the font size for the Nebula Pad

ajs123, 2024-03-12:
In main.cpp lines 295-298, the default font is set to 12 or 16 pixels high according to the vertical resolution of the screen. For reasons I've not discerned, that setting is not overriding the default in lv_conf.h. The intent, presumably, is to minimize editing of source files necessary for building binaries for different platforms. As a temporary fix, editing the default font in lv_conf.h will provide the 12-pixel font:

```
#define LV_FONT_DEFAULT &lv_font_montserrat_12
```
at line 371.

spitfire, 2024-03-12:
Hey, if I substitute that value (and correct the calibration as per #54) how can I compile and install it on 3 V3 KE (I already have the base version installed)?

Liionboy, 2024-03-25:
Create a PR with this to be implement in next version

closed 2024-04-14 by ballaswag