ballaswag/guppyscreen #59: stuck on waiting for printer to initialize

keysCodes, 2024-03-19:


keysCodes, 2024-03-19:
just on the top been there for a bit turned it off and everything

keysCodes, 2024-03-19:
ENDER 3 V3 SE WITH NEBULA PAD BTW THAT MIGHT BE THE ISSUE

ballaswag, 2024-03-20:
Make sure moonraker is working. It’s require for any clients like guppy screen, klipper screen, fluidd and etc.

adec, 2024-03-21:
@keysCodes have you got the spoolman configured in moonraker by any chance?  If so there is a separate issue open for this. See issue #49.

keysCodes, 2024-03-23:
no is spoolman a external program that doesnt come with guppyscreen because with the ender 3 v3 se without a rasberry pi you cant install certain things u can only root it with a nebula pad and install certain things

keysCodes, 2024-03-23:
> Make sure moonraker is working. It’s require for any clients like guppy screen, klipper screen, fluidd and etc.

i think it is because fluidd and mainsail work fine