ballaswag/guppyscreen #65: Ender 3 V3 KE : several non functional entries

davphoenix31, 2024-04-01:
On ender 3 V3 KE:
- calibration of vibration is not working (logs about connexion issue with sensor whereas working with original creality menu)
- extrusion is not working (extrusion is never ending, obliged to power off)
Happy to contribute/test again when functional for ender 3 V3 KE

henrivdr, 2024-04-02:
I have also noticed that the filament load extrudes excessively long on the CR10-SE.

ballaswag, 2024-04-02:
@henrivdr the unload and unload button are mapped to default macros provided by creality. You can customize them to what you like or remap them to your own macros. The mapping is in the /usr/data/guppyscreen/guppyconfig.json file.

@davphoenix31 what’s the calibration of vibration? Are you referring to input shaping, belt tension?

davphoenix31, 2024-04-02:
Hello @ballaswag 
I am referring to ADXL345 Vibration Compensation Sensor

davphoenix31, 2024-04-02:
@ballaswag 
For extrusion, I clicked and the filament was extruded continuously