ballaswag/guppyscreen #73: Onscreen Keyboard Multiple touch registers

JustBlan1984, 2024-04-12:
When trying to use the On Screen Keyboard on the Creality K1 Max and using Guppy the keyboard registers 3 touches for 1 touch on any key.

ballaswag, 2024-04-14:
Can you share a video of this behavior? I have not been able to reproduce it.

davphoenix31, 2024-04-15:
For info, works very well on KE and screen responsiveness is significantly better after.

natesaninja, 2024-04-22:
I’m having the same issue on both of my K1 maxes. I have to hit it lightly really quick

natesaninja, 2024-04-22:
https://github.com/ballaswag/guppyscreen/assets/38551021/c5286db4-26c3-4702-b3f5-95cef4ca95ef

natesaninja, 2024-04-22:
Nvm I just update Guppy from on screen and the problem has gone away

evgarthub, 2024-04-25:
I'm experiencing  the same issue, now I can't adjust temperature due to it.

evgarthub, 2024-04-25:
I'm using latest 23-beta, I originally installed guppyscreen by Creality-Helper-Scripts but then re-installed it directly from this repo and issue is not gone.

ballaswag, 2024-04-25:
I pulled the latest version that fixed this issue but introduced a more serious bug where print status gets stuck. If you download 0.0.21-beta from releases and overwrite your guppyscreen dir. It will fix the issue.

I have a new release planned with all the fixes, but haven't have time to finish testing it yet. Hope to get it out this week.

ballaswag, 2024-04-26:
0.0.25-beta is out, fixes all the things and has  better speed/flow/layer count logic too.

closed 2024-04-26 by ballaswag