ballaswag/guppyscreen #74: Ender 3 V3 KE  -- Display Sleep Problem

Liionboy, 2024-04-12:
Hello,

There are a problem with selecting Display Sleep on Nebula Pad , you can select only 5 hours , can't select Never or something else, i try all posibility.

Thanks

ballaswag, 2024-04-14:
The nightly build should fix this, will be in next stable release as well. For now you can edit guppyconfig.json `display_sleep_sec` to -1 for Never.

davphoenix31, 2024-04-15:
Yes, I do confirm the same for me on KE. Additionally, it seems the default value is low, something like 30s whereas the first value on the drop-down is 10min

ballaswag, 2024-04-15:
Just released 0.0.23-beta, please update and see if issue is resolved.

davphoenix31, 2024-04-15:
Yes, I do confirm, working well with 0.0.23-beta, well done.

closed 2024-04-15 by ballaswag