ballaswag/guppyscreen #76: Touchscreen keypad not debounced?

MrDix, 2024-04-14:
On my K1 if I try to enter a temperature via the guppy on screen keypad all numbers show up 2-3 times. Im not fast enough to enter singe digit. E.g. this means I can not set the temperature of the bed to 100°C. The closest I can set it is 99°C or 111°C. 
Regular menu buttons or file chooser touch screen works flawless.
I have installed the newest version of guppy screen (updated via the on screen update).

I assume the others do not face this issue as someone would have complained already. Is there a option to active debouncing for the keypad which I'm not aware of, or can I have a different problem?

Thank you for the hard work and all of your awesome tools you have created for the K1!

Kind regards,
Th0mas

ballaswag, 2024-04-14:
Please share a short video showing the behavior. Will take a look.

MrDix, 2024-04-14:
Currently its printing but as soon as it is done I'll shoot the video. Thank you for the quick reply!

MrDix, 2024-04-14:
Here is a short video which shows the problem on my Creality K1 with guppy screen 0.0.22-beta (and previous versions):
https://youtu.be/K0vR7zI4lQ8?si=K-fFl_c5kAdxBMX3

ballaswag, 2024-04-15:
This is new to me. This happens even before 0.0.22-beta? Give latest 0.0.23-beta a try (nothing specific was added to handle this behavior though).

MrDix, 2024-04-15:
It was happening with my previous version as well but I can not tell you if it worked in an (much) earlier version as I have never typed temperatures on the screen before. I've updated to 0.0.23-beta an the problem persisted.

ajs123, 2024-04-15:
I'm chiming in because I helped with a pulldown menu issue that _could_ be related. @MrDix - If you go to Options (the gear) then System, do the switches for the last two options seem to work properly? When you tap once as you would on the numeric keypad, does a switch ever go off then immediately back on?

I have no experience with lvgl, but I'm wondering whether some of these resistive touchpads are bouncier than others.

MrDix, 2024-04-16:
@ajs123 Thank you for joining in. The two switches in the System menu work flawless and have no bouncing problem on my screen like the keypad has. Have clicked several times on each of them and they always switched like expected.

chiksum, 2024-04-16:
Yes this happened to me too, I am on 0.0.23-beta.
The keyboard is unable to use.

mxfi, 2024-04-16:
Adding in a +1 in terms of having the same multi tap, normal taps come out as 2-4 items (ie selecting 1 give 1111). When touch tapping it as fast as I can, I can make it give a single digit. This is new after updating guppy screen and helper script. I think when running last firmware (or whatever it was 2 weeks ago, I never had this issue, only after updating recently). I also noticed a bit of a burn in on the screen which I never experienced before, don't know if it's related.

Running 1.3.3.8 Creality and most recent guppy + helper script

ajs123, 2024-04-16:
@ballaswag [this](https://patchwork.kernel.org/project/linux-arm-kernel/patch/20170315155140.54518-4-icenowy@aosc.xyz/) might be the kernel module behind /dev/input/event0. If so, it's simply reading the current state every 30 ms. If you think that a debounce in lv_tc_indev_drv_read_cb.c is appropriate, I don't mind writing one. I do not have this problem on my Nebula pad but it should be testable by setting a long debounce interval.

ballaswag, 2024-04-16:
> @ballaswag [this](https://patchwork.kernel.org/project/linux-arm-kernel/patch/20170315155140.54518-4-icenowy@aosc.xyz/) might be the kernel module behind /dev/input/event0. If so, it's simply [...]

I'm not sure debouncing is the actual issue. I'm running the latest on my K1 as well, but I can't reproduce the problem. The K1 path doesn't get wrapped by the lv_tc* stuff because it's calibrated as is. It directly interacts with evdev_read. The only thing that has changed was removing the coordinate clamping.

If anyone with this issue, please install 0.0.21-beta and see if the issue was present before.

MrDix, 2024-04-17:
How can I downgrade guppyscreen? I've downloaded guppyscreen.tar.gz from assets list of release 0.0.21-beta, copied it to the printer, extracted it an run the guppyscreen/installer.sh but this installer uses curl to download and install the latest version.

ballaswag, 2024-04-17:
Download guppyscreen.tar.gz here https://github.com/ballaswag/guppyscreen/releases/tag/0.0.21-beta

Stop guppyscreen:
/etc/init.d/S99guppyscreen stop

tar xf guppyscreen.tar.gz
Copy and replace guppyscreen with the new one. The restart it:

/etc/init.d/S99guppyscreen restart

MrDix, 2024-04-17:
In version 0.0.21-beta the keypad is working flawless: Single click on any number buttons result in singe digits.

ballaswag, 2024-04-19:
Nightly https://github.com/ballaswag/guppyscreen/releases/tag/nightly fixes the keyboard issue.

ballaswag, 2024-04-19:
Nightly is the latest development build. The fix is simple, I accidentally removed the indev driver initialization for the non-calibrated path:

https://github.com/ballaswag/guppyscreen/commit/97de810936e2628d57154f6db555b5bbf494656a

mxfi, 2024-04-20:
> Nightly is the latest development build. The fix is simple, I accidentally removed the indev driver initialization for the non-calibrated path:
> 
> [97de810](https://github.com/ballaswag/guppyscreen/commit/97de810936e2628d57154f6db555b5bbf494656a)

Ah, still new to git and code, thanks for the fix and explaining it :)

downloaded newest beta 0.0.24 and it's working great!

Cheers

MrDix, 2024-04-22:
Thank you very much and sorry for the late reply! I was away from home for three days due to an injury.

closed 2024-04-22 by MrDix