ballaswag/guppyscreen #80: Screen enters sleep haphazardly on the V3 KE

Orel-A, 2024-04-15:
The screen will turn off randomly without respecting the 10 minutes idle time set on the V3 KE.
However when setting to Never in the settings, it doesn't happen. So maybe, something is wrong with the timer?

Orel-A, 2024-04-16:
Anyone knows where is the logs directory?

ballaswag, 2024-04-16:
I observed the same, approximately every hour it will wake up and sleeps again even when there's no touch activity. Will take a look.

Orel-A, 2024-04-18:
Have you found anything? Is there a way to debug this on my end?
Maybe a debug version I can install so it will generate a stack trace?

henrivdr, 2024-07-17:
On the CR10 SE (Nebula pad) the screen always blanks a few seconds after startup.
It's exactly at the point where I start using it. And it catches me by surprise, **every** time 😬