ballaswag/guppyscreen #81: Error when running inputshaping

Brakisshuro, 2024-04-16:
So it completes but when it goes to output my graph;

12:30:22 PM 
!! ({'code': 'key314', 'msg': "Error writing to file '%s': %s", 'values': ['%s', '%s']}, '/tmp/calibration_data_x_20240417_032250.csv', '[Errno 28] No space left on device', '/tmp/calibration_data_x_20240417_032250.csv', '[Errno 28] No space left on device')
12:30:22 PM 
!! ({'code': 'key314', 'msg': "Error writing to file '%s': %s", 'values': ['%s', '%s']}, '/tmp/calibration_data_x_20240417_032250.csv', '[Errno 28] No space left on device', '/tmp/calibration_data_x_20240417_032250.csv', '[Errno 28] No space left on device')

I have 3 GB available on printer so not sure what the issue is. This is only on one of my 4 installs of this on my K1M's

Brakisshuro, 2024-04-29:
Hello?

ballaswag, 2024-04-29:
What input shaping macros are you running? Guppy ones don't have the unix timestamp in their file names so that each run replaces the existing ones.

/tmp is small, not like the data partition. So if you run non-guppy shapers, they default to create a new sets of file each run. Try restarting your printer and use the guppy input shaping macros.

Brakisshuro, 2024-04-29:
I thought I was running the guppy ones. I will look again. Anyway we can get confirmation of the buttons we press? Loading and unloading is very frustrating