ballaswag/guppyscreen #85: stuck with non existing jon in Queue

avilshadow, 2024-04-21:
guppu screen 0.0.24-beta
printerv3KE

Used 0.0.21beta before, updated with the button on the screen, not when trying to start new print got this screen on the screenshot.
QUwuw/ View job buttons are also non-functional.
Could start prijnt with web UI..

[image]

Strahler85, 2024-04-21:
I have the same issue with beta 0.24. I can start prints over fluiid and OrcaSlicer.

natesaninja, 2024-04-23:
Same issue as well

Teense, 2024-04-23:
Same issue, KE version 24.

ballaswag, 2024-04-24:
Found the issue. Will make a release soon.

ballaswag, 2024-04-26:
0.0.25-beta is out, fixes all the things and has  better speed/flow/layer count logic too.

closed 2024-04-26 by ballaswag