ballaswag/guppyscreen #86: Revert to a previous verions

Liionboy, 2024-04-22:
Hello,

How can i revert back to version 23 because 24 is full of bug at display:

[image]

Thanks

Liionboy, 2024-04-22:
No speed, no thumbnail, no correct Layers, Time Rmaining not work

ballaswag, 2024-04-22:
Rerun the installer, I pulled 0.0.24-beta. Latest should be reverted to 0.0.23-beta now.

Liionboy, 2024-04-22:
I did and still is 0.0.24

Thanks

Liionboy, 2024-04-22:
Oh with update don't work, i run installer and worked.

Thanks

martinrevert, 2024-04-22:
Sorry for the offtopic @Liionboy , How did you phisically achieve to rotate the Nebula pad? Did you printed a custom support to get the screen horizontal? If so, do you have the STL?. Thanks!

Liionboy, 2024-04-22:
Hello,

You find here

https://www.printables.com/model/706657-creality-ender-3-v3-e3v3-se-ke-and-cr-10-se-portra

ballaswag, 2024-04-26:
0.0.25-beta is out, fixes all the things and has  better speed/flow/layer count logic too.

closed 2024-04-26 by ballaswag