ballaswag/guppyscreen #87: Numbers repeat rapidly when setting temperature

dprossner, 2024-04-24:
when entering numbers to set temperatures on the keypad that pops up, there is no delay before numbers start repeating so entering 200 degrees at a reasonable speed results in 2222220000000000000000. I suggest repeating numbers be disabled altogether when inputting temperatures so one press regardless of duration enters one character.

MrDix, 2024-04-25:
Looks like a duplicate of https://github.com/ballaswag/guppyscreen/issues/76

This was a bug in 0.0.22-beta and 0.0.23-beta and should have been fixed in guppyscreen 0.0.24-beta. Are you using the current version?

evgarthub, 2024-04-25:
Also duplicate of #73

dprossner, 2024-04-26:
my mistake. on me for assuming my helper script installed the latest version

closed 2024-04-26 by dprossner