ballaswag/guppyscreen #90: Hot-end jam/clog on Ender 3 V3 KE using GuppyScreen's load/unload filament buttons

grantland, 2024-05-04:
Since updating GuppyScreen I've been getting hot-end jam/clogs when using the load/unload filament buttons. I've tried using `LOAD_MATERIAL` and `QUIT_MATERIAL` macros instead and am able to change filament without jams/clogs.

Due to this, it seems like the issue is due to the following commit which changes GuppyScreen's load/unload filament buttons from `LOAD_MATERIAL`/`QUIT_MATERIAL` to `_GUPPY_LOAD_MATERIAL`/`_GUPPY_QUIT_MATERIAL`: https://github.com/ballaswag/guppyscreen/commit/e21b16320801ee26f16dfdb2364d5081d5f3f4e6

ballaswag, 2024-05-06:
People were complaining about the default amount of filament extruded for those defaults. You can always update your guppyconfig.json to remap these button to the old macros, see doc here https://ballaswag.github.io/docs/guppyscreen/configuration/#load_filament

Will look to revert them if clogging is an issue.

grantland, 2024-05-06:
FWIW I got hotend jams/clogs 2x in a row when using `_GUPPY_LOAD_MATERIAL`/`_GUPPY_QUIT_MATERIAL` and they have since stopped after switching back to `LOAD_MATERIAL`/`QUIT_MATERIAL`.

The link to the config is greatly appreciated and I'll be switching the configs to `LOAD_MATERIAL`/`QUIT_MATERIAL`. Thanks!

Orel-A, 2024-05-08:
These are the macros in the V3 KE config files:
```
[gcode_macro LOAD_MATERIAL]
gcode:
  SAVE_GCODE_STATE NAME=myMoveState
  M109 S{printer.custom_macro.default_extruder_temp}
  G91
  G1 E150 F180
  RESTORE_GCODE_STATE NAME=myMoveState

[gcode_macro QUIT_MATERIAL]
gcode:
  SAVE_GCODE_STATE NAME=myMoveState
  M109 S{printer.custom_macro.default_extruder_temp}
  G91
  G1 E20 F180
  G1 E-30 F180
  G1 E-50 F2000
  RESTORE_GCODE_STATE NAME=myMoveState
```

I only noticed that _GUPPY_LOAD_MATERIAL is missing the save gcode state and that the fan2 macros are giving errors in the console since they don't exist on the V3 KE. But the guppy macro is working fine most of the time.
Today I experienced something similar to what @grantland said, I cancelled a print and right after I tried to Unload and Load. For some reason the extruder didn't move at all. Only after restarting the machine I could.

grantland, 2024-05-21:
FWIW my `guppyconfig.json` somehow got reverted and caused a clog again. Seems like changing `guppyconfig.json` like the following persists:

```
[gcode_macro _GUPPY_LOAD_MATERIAL]
gcode:
  LOAD_MATERIAL

[gcode_macro _GUPPY_QUIT_MATERIAL]
gcode:
  QUIT_MATERIAL
```