ballaswag/guppyscreen #96: Its just really really bad on ender 3 v3

IdontWannaBeANoob, 2024-05-12:
Everything about this mod has me waiting on my print to finish so I can uninstall it. 
Extremely slow response time to touch input, if any. 
No z offset and bed mesh auto start buttons that a company like creality can even include on their machines. 
The third tab down that I can't name because it's frozen on the tmc metrics page during a print is formatted poorly. I can't efficiently scroll commands without the keyboard opening and half of the screen is wasted. I can't wait to remove this low effort software from my printer once/if this print finishes.
I think the only worse thing I could do for my printer would be to put it in the dumpster full of napalm with a road flare to match where this software belongs. Just stop trying, don't even bother fixing it. It's worse than stock creality. Let me say that again to drive the point home. CREALITY WRITES BETTER CODE AND MAKES BETTER SCREEN LAYOUT AND THEY ARE A JOKE IN THE COMMUNITY. 
Go watch the YouTube videos complaining about this programming fail.

Teense, 2024-05-27:
No one made a guppyscreen for Ender 3 v3 lol, calm down.

IdontWannaBeANoob, 2024-05-27:
> No one made a guppyscreen for Ender 3 v3 lol, calm down.

https://github.com/Guilouz/Creality-Helper-Script

It's literally right in the installation helper. But it's been updated and works pretty good now.

IdontWannaBeANoob, 2024-05-30:
Does the se come with the nebula screen?

DarthRidar, 2024-05-30:
No. I have the Creality Nebula Smart Kit.