ballaswag/guppyscreen #99: K1 Max with Guilouz script - No module named 'numpy'

nmh5028, 2024-05-23:
Receive this error when running input shaper from the on screen buttons. Clean install on K1 max using Guilouz script, "calibrate" button with "graph" selected. I have attempted a full factory reset and have the same issue

`22:52:59 
// Wait for calculations..
22:53:03 
// calc_freq_response return (0)
22:53:03 
// Resonances data written to /tmp/resonances_y_y.csv file
22:53:03 
// Running Command {guppy_input_shaper}...:
22:53:04 
// Traceback (most recent call last):
// File "/usr/data/printer_data/config/GuppyScreen/scripts/calibrate_shaper.py", line 12, in
22:53:04 
// import numpy as np, matplotlib
// ModuleNotFoundError: No module named 'numpy'
22:53:04 
// Command {guppy_input_shaper} finished
22:53:04 
// Running Command {guppy_input_shaper}...:
22:53:05 
// Traceback (most recent call last):
// File "/usr/data/printer_data/config/GuppyScreen/scripts/calibrate_shaper.py", line 12, in
22:53:05 
// import numpy as np, matplotlib
// ModuleNotFoundError: No module named 'numpy'
22:53:05 
// Command {guppy_input_shaper} finished
`

closed 2024-05-23 by nmh5028

AlikaTheDog, 2024-09-24:
@nmh5028 How did you solve this issue?

damian5602, 2024-10-13:
Any solution for that?
pip install numpy does not works, getting some errors

nepvee, 2025-01-25:
I remember solving it in the past, however, I ended up having to reset my printer at some point and cannot seem to remember how… Anyone here have any solutions?

nepvee, 2025-01-25:
I installed python3-dev using entware, and now the input shaper stuff works fine, if anyone comes across this problem they could try that.