ballaswag/guppyscreen #102: Calibration not working with inverted screen (display_rotate:2)

bferrarese, 2024-06-06:
I'm using a V3 KE with the pad upside down. 
I have the display_rotate set to 2 and it was working as expected. (can't recall version I was using before)
After an update the touch input is upside down (related to my view - ie back regular view).
If I want to click setting I have touch the upper/right corner instead of the bottom/left corner. 
Calibration was done with screen upside down (display_rotate:0) so I tried to force a new calibration with the screen on its correct orientation (display_rotate:2) but still the same behavior. 
It feels like the screen was flipped but not the touch.

coreflake1, 2024-06-07:
I'm having the same issue. 
Screen rotated 180 degrees but calibration is inverted.

shadowpanther, 2024-07-07:
Same with 1 and 3 rotations - calibration is not rotated, only the picture is.

bobiaba, 2024-11-06:
Same issue. Tried swapping minuses in `touch_calibration_coeff` in different ways with no avail

bferrarese, 2024-11-26:
I have upload the guppyscreen binary for those who need.

https://github.com/bferrarese/guppyscreen-tata/releases/download/MIPS/guppyscreen.