ballaswag/guppyscreen #111: Is the project no longer updated?

poli27, 2024-07-21:
Will there be an update or has the project stopped?

Brakisshuro, 2024-08-13:
seconded

boromyr, 2024-09-04:
Has it been interrupted? Is there any official confirmation?

Atomique13, 2024-10-06:
👀

bhmyers66, 2024-11-12:
Bueller? Bueller?

bhmyers66, 2024-11-13:
@ballaswag will there be any more updates to Guppyscreen?

ballaswag, 2024-11-13:
I don't have any immediate plans for guppy screen. If you guys have any specific features or urgent bug fixes needed, I will take a look as time allows.

Brakisshuro, 2024-11-13:
> I don't have any immediate plans for guppy screen. If you guys have any specific features or urgent bug fixes needed, I will take a look as time allows.

There were a couple of issues in the issues area, maybe line those up and release a final?

bhmyers66, 2024-11-14:
> I don't have any immediate plans for guppy screen. If you guys have any specific features or urgent bug fixes needed, I will take a look as time allows.

I second the motion of a final release. Maybe someone else with take up LTS?

Tronutopia, 2025-05-15:
I would love to see it progress out of Beta into a full fledged version.

Tronutopia, 2025-05-15:
A couple of issues that I have...
1. When I start my printer (Creality Ender-3 V3 SE with Nebula Pad), the Guppy UI does NOT display. The Creality logo is all I see.
But when I start Creality Print/Fluidd then the Guppy UI will display. It never used to do this, it would always start up okay in a matter of seconds.
2. I have the screen time set for 10 mins but it seems very intermittent. Sometimes the screen will stay on for hours and other times for 10 mins.

Tronutopia, 2025-05-16:
How do I get the print thumbnail?
Mine only shows the horizontal bar with percentage complete.
But no thumbnail.

Helix751, 2026-01-04:
> [@ballaswag](https://github.com/ballaswag) How do I get the print thumbnail? Mine only shows the horizontal bar with percentage complete. But no thumbnail.

Same here.