ballaswag/guppyscreen #117: Request: Material Unload Temp to not override a manual temp when higher than 240*

Marine6680, 2024-11-19:
I am printing with more engineering materials on my K1 Max.

When using the material load/unload commands under the extrude menu screen, the temperature changes to 240* no matter what temperature I have set manually.

I would like the option to either select the temperature on the extrude screen or for the load/unload command to not override my manually set temperature if it is currently set higher than 240*, as I need 300* for some of my filaments to load/unload properly.

If not a change to the load/unload macro tied to the buttons on the screen, I would like the temperature select buttons to also include 300*, so I can use the manual extrude/retract function. If there is no room to add an additional selection, I would suspect that many people do not need the lowest temperature option offered.

I would also appreciate the ability to monitor the chamber temp on the Print Progress screen. It would be useful for ensuring the temps are staying steady when printing engineering materials.

Thanks

Marine6680, 2025-01-08:
I found a way to do this by modifying the guppyscreen cfg file.

This change will do the following:
If the extruder temp s already above 240* the macro will just start the extrusion
If the temp is below 240* it will set the extruder temp to 240* and wait for temp, then extrude.

Limitations, the macro does not know the extruder set temp, so for example, if it is set to 300* but is currently at 250*, the macro will start the extrusion without waiting for the extruder to reach its current set point first.

I am unaware of how to write the macro to do that. It took a while to figure out how to get it this far.

Another thing I attempted was to have the screen give a popup message that it was loading or unloading filament. All I managed was to get a message to show in the console.

Here is the modified macro code for those interested.

[gcode_macro _GUPPY_LOAD_MATERIAL]
gcode:
  {% if printer['extruder'].temperature > 240|int  %}
   {% set extrude_len = params.EXTRUDE_LEN|default(60)|int %}
   RESPOND MSG="Loading Filament"
   G91
   G1 E{extrude_len} F180
  {% else %}
   {% set extruder_temp = params.EXTRUDER_TEMP|default(240)|int %}
   {% set extrude_len = params.EXTRUDE_LEN|default(60)|int %}
   RESPOND MSG="Loading Filament"
   LOAD_MATERIAL_CLOSE_FAN2
   M109 S{extruder_temp}
   G91
   G1 E{extrude_len} F180
   LOAD_MATERIAL_RESTORE_FAN2 # k1 stuff
  {% endif %}

[gcode_macro _GUPPY_QUIT_MATERIAL]
gcode:
  {% if printer['extruder'].temperature > 240|int  %}
   SAVE_GCODE_STATE NAME=myMoveState
   RESPOND MSG="Unloading Filament"
   G91
   G1 E20 F180
   G1 E-30 F180
   G1 E-50 F2000
   RESTORE_GCODE_STATE NAME=myMoveState
  {% else %}
   {% set extruder_temp = params.EXTRUDER_TEMP|default(240)|int %}
   SAVE_GCODE_STATE NAME=myMoveState
   RESPOND MSG="Unloading Filament"
   M109 S{extruder_temp}
   G91
   G1 E20 F180
   G1 E-30 F180
   G1 E-50 F2000
   RESTORE_GCODE_STATE NAME=myMoveState
  {% endif %}

koenpunt, 2025-02-15:
FYI: I've created a PR that should allow higher temperatures to be selected; #136

koenpunt, 2025-02-22:
And I've also created a PR to add the chamber temperature to the print status screen; https://github.com/ballaswag/guppyscreen/pull/141