ballaswag/guppyscreen #119: Android app crashes on launch

living-relation, 2024-11-21:
Installed from your apk. With an error "app has stopped due to a bug". I'm using Samsung Galaxy S10. 9 times out of 10 it crashes. When it does open, it crashes when i try to do anything.

sudokit, 2024-12-13:
Is the source code available somewhere for the app?

ballaswag, 2024-12-13:
Try the non nightly maybe? The android code is under the android branch for anyone wants to take a look. Build processes are all in the github workflow.

living-relation, 2024-12-14:
There's only the one apk download link on the main page? Where should i be looking for nightly android builds?

ballaswag, 2024-12-14:
Check here https://github.com/ballaswag/guppyscreen/releases

No guarantee they'll work. I haven't tested them in a while.

sudokit, 2024-12-14:
The latest one from the android branch seems to be crashing as well, the non-nightly 0.0.25-beta seems to work just fine, but doesnt work in horizontal orientation, so im guessing somethings happens in the swapchain or graphics buffer idk how it works, and it becomes invalid and crashes